OpenIPC/msposd #4: Move font files from `bin` folder

seriyps, 2024-09-27:
I find it strange that font files are expected to be in `/usr/bin`. This directory is intended to be used for executables only.

Makes more sense to put them somewhere in `/usr`

https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard

tipoman9, 2024-09-28:
The initial install option is simple bulk copy all to some folder. And since I considered ```/usr/bin/`` is a good place for the app, I hardcoded the path for the fonts just to be easier for me to test, and after that to explain how to install
When msposd makes it to OpenIPC firmware, the location of font files can easily be changed or maybe the app should check several folders?

tipoman9, 2024-10-18:
The default location for Flight Controller's png "font" files is changed to ```/usr/share/fonts```

closed 2024-10-18 by tipoman9