OpenIPC/msposd #12: RGN_GetAttr failed with 0xa0032001 -- osd icons error

lida2003, 2024-10-27:
```
$ msposd --master /dev/ttyS2 --baudrate 115200 -c 7 -c 9 --out 127.0.0.1:14555 -osd -r 20 --ahi 0 --wait 5 --persist 50 -v
Ver: 20241019_230853
Monitoring RC channel 7
Monitoring RC channel 9
MSP to OSD mode!Verbose mode!Majestic width:1280,height:720
Loading /usr/share/fonts/font_hd.png for 720p mode
I4 48:9216 Stride:24
Font file res 48:9216 pages:2
Glyph size:24:36 on a 53:20 matrix. Overlay 1280:720
MI_RGN_Init results: 0
[create_region:64]RGN_GetAttr failed with 0xa0032001 , creating region 6...
[create_region:112]RGN_GetDisplayAttr failed with 0xa0032012 6, attaching...
Create_region PixelFormat:3 Size: 1280:720 X_Offset:0 results: 0
Set SS Font Review 1280:720USING DIRECT VIDEO MEM MODE!!!
set_LOGO with u32Height:720 enPixelFormat 3
Listening on /dev/ttyS2...
UART Events:0 MessagesTTL:0 AttitMSGs:0(0ms) Bytes/S:0 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
$M<Flight Controller detected: ARDU
UART Events:5 MessagesTTL:6 AttitMSGs:0(0ms) Bytes/S:80 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
UART Events:4 MessagesTTL:6 AttitMSGs:0(0ms) Bytes/S:83 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
```
[image]

lida2003, 2024-10-27:
tried latest code on git, still same result:

```
# msposd --master /dev/ttyS2 --baudrate 115200 -c 7 -c 9 --out 127.0.0.1:14555 -osd -r 20 --ahi 0 --wait 5
 --persist 50 -v
Ver: 20241027_153531
Monitoring RC channel 7
Monitoring RC channel 9
MSP to OSD mode!Verbose mode!Majestic width:1280,height:720
Loading /usr/share/fonts/font_hd.png for 720p mode
I4 48:9216 Stride:24
Font file res 48:9216 pages:2
Glyph size:24:36 on a 53:20 matrix. Overlay 1280:720
MI_RGN_Init results: 0
[create_region:64]RGN_GetAttr failed with 0xa0032001 , creating region 9...
[create_region:112]RGN_GetDisplayAttr failed with 0xa0032012 9, attaching...
Create_region PixelFormat:3 Size: 1280:720 X_Offset:0 results: 0
Set SS Font Review 1280:720USING DIRECT VIDEO MEM MODE!!!
set_LOGO with u32Height:720 enPixelFormat 3
Listening on /dev/ttyS2...
UART Events:0 MessagesTTL:0 AttitMSGs:0(0ms) Bytes/S:0 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
Flight Controller detected: BTFL
PuTTYUART Events:6 MessagesTTL:8 AttitMSGs:0(0ms) Bytes/S:117 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
UART Events:6 MessagesTTL:6 AttitMSGs:0(0ms) Bytes/S:106 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
UART Events:5 MessagesTTL:6 AttitMSGs:0(0ms) Bytes/S:82 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
UART Events:4 MessagesTTL:7 AttitMSGs:0(0ms) Bytes/S:108 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
UART Events:5 MessagesTTL:7 AttitMSGs:0(0ms) Bytes/S:108 FPS:1(skipped:0), avg time per frame ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
```

tipoman9, 2024-10-27:
```[create_region:64]RGN_GetAttr failed with 0xa0032001 , creating region 6...``` is not a critical error.
There is no data on your uart port.
_Check your wiring. You need Tx/Rx/Gnd wires. Data lines must be crossed (Rx goes to Tx ).
Check Inav/Ardu config. Enable MSP and OSD where needed! There is a separate option for OSD in INAV. Do not enable mavlink and msp on a single UART. (This is a bug in BF). Do not start msposd twice._

lida2003, 2024-10-27:
OK. I re-configed and my requirement is a little bit special.

I have two cameras: 1 for analog; 1 for openipc.

Now I need to keep analog settings, which is default:

- MP Onboard configure icon location
```
SERIAL3_PROTOCOL = 44  //IRC_TRAMP
SERIAL3_OPTIONS = 4 //halfDuplex
OSD_TYPE = 1 //MAX7456
```

- Do I need to configure icon location on MP??? it seems useless and it's NOT realted with msposd
Is there any settings I should configure? or something I have configured wrong? As there is no such guide avaliable, I just configure by sense. 

https://ardupilot.org/plane/docs/common-msp-osd-overview.html

```
SERIAL8_BAUD = 115 // 115200
SERIAL8_PROTOCOL = 42  //DISPLAYPORT
SERIAL8_OPTIONS = 0 //polling
OSD_TYPE2 = 5 //MSP_DISPLAYPORT
```

[image]
[image]

tipoman9, 2024-10-27:
MSPOSD canvas size is 53x20 characters, it seems from the picture above that you have chosen a profile with lower dimensions.

lida2003, 2024-10-27:
> On lower-end cameras like gk7205v200/v210 the OSD will work only in 1280x720 mode!

I'm using 720p, any default settings i should change? As my goggles fatshark hd2 do not have 1080p resolution.

lida2003, 2024-10-27:
The poblem is the icon is on the left up corner. I hope it should be adjusted as below:

[image]

lida2003, 2024-10-27:
I tried 1080P, still up left corner

[image]

lida2003, 2024-10-28:
Where are those icon x-y coordinates definition?

lida2003, 2024-10-28:
Close it for further questions:

- [Is it possible for two OSD resolution working at the same time?](https://discuss.ardupilot.org/t/is-it-possible-for-two-osd-resolution-working-at-the-same-time/125505)
- [ OSD Icon miss matched #13 ](https://github.com/OpenIPC/msposd/issues/13)

closed 2024-10-28 by lida2003