OpenIPC/msposd #13: OSD Icon miss matched

lida2003, 2024-10-28:
It's different from OSD tab, any idea?

[image]

[image]

lida2003, 2024-10-28:
OK, I figured out. msposd do NOT have ardupilot png files.

- https://sneaky-fpv.gitbook.io/hd-osd-and-tools

[image]

tipoman9, 2024-10-28:
Most users reported they can use Betaflight fonts when setting ```MSP_OPTIONS fonts = 4``` in ardupilot.
Can you summarize what config changes are needed to make ArduPilot use its native font?

lida2003, 2024-10-28:
**Analog Camera:**

```
SERIAL3_BAUD = 230     //230400
SERIAL3_PROTOCOL = 44  //IRC_TRAMP
SERIAL3_OPTIONS = 4    //halfDuplex
OSD_TYPE1 = 1          //MAX7456
```

**Digital Camera:**

```
SERIAL8_BAUD = 115     //115200
SERIAL8_PROTOCOL = 42  //DISPLAYPORT
SERIAL8_OPTIONS = 0    //polling
OSD_TYPE2 = 5          //MSP_DISPLAYPORT
MSP_OPTIONS = 4        //EnableBTFLFonts
```

Above settings for two camera works, and I think a more convient local postion feature is much more handy in my case. But I think there are not that many people use. Anyway, I have post a request here. 

- [ OSD icon local position configuration support #14 ](https://github.com/OpenIPC/msposd/issues/14)

tipoman9, 2024-10-28:
So the above options work with Betaflight fonts, right?
What settings are needed to make ardupilot use its native fonts ?

lida2003, 2024-10-28:
> What settings are needed to make ardupilot use its native fonts ?

I can't make it work, even the initial icons are displayed as garbled image. And there is a bug I think https://github.com/OpenIPC/msposd/issues/15

tipoman9, 2024-10-28:
Well, several users reported ardupilot latest version working with Betaflight fonts with these settings:
```OSD_TYPE=5
SERIALx_PROTOCOL=42
MSP_OPTIONS=6```

lida2003, 2024-10-28:
I don't have any idea about MSP_OPTIONS=6. Did you get feedback from developer or users, as I couldn't find any links about OpenIPC configuration, especially those unused settings.

[image]

tipoman9, 2024-10-30:
As I explained you here, https://github.com/OpenIPC/msposd/issues/15
Ardupilot can be configured to use Betaflight fonts.
If you have trouble with this, please ask in Ardupilot channel for help.

closed 2024-10-30 by tipoman9

lida2003, 2024-10-30:
> Ardupilot can be configured to use Betaflight fonts.

No, msposd doesn't support native ardupilot display port. As the incon index is different. Then you use a betaflight configuration option, but the power icon is wrong . Then you said it's png icon index is different fron betaflight. 

So I'm NOT sure if it's png issue or msposd code issue or ardupilot's code issue on msp option. Anyway, it didn't work with that icon, am i right?

tipoman9, 2024-10-31:
_Currently msposd is not tested with ardupilots "native" fonts, since the demand for this was lacking.
Several users reported that they can configure ardupilot to output osd icon index for Betaflight fonts.
I hope I'll be able to add support for native Ardu fonts when I find a spare flight controller to install it and test it. (apart from the two I use to test Inav and Betaflight)_

There is no "VTX power icon" in Betaflight fonts, so when you configure Ardupilot to use Betaflight fonts it simply takes the glyph from the location it is supposed to be in Ardu fonts I suppose.
Since this icon is so important for you, you can manually copy it in the betaflight font png.