OpenIPC/msposd #15: betaflight png `mw` icon is NOT good!

lida2003, 2024-10-28:
Down right should be 25mW.  The icon is not good.

[image]

tipoman9, 2024-10-28:
Could you please provide more details when posting bug reports?

closed 2024-10-28 by tipoman9

tipoman9, 2024-10-29:
@lida2003 
Currently msposd is not tested with ardupilots "native" fonts, since the demand for this was lacking.
Several users reported that they can configure ardupilot to output  osd icon index for Betaflight fonts.
From the sample above seems your ardupilot setup is trying to use its "native" fonts instead.
[image]

So please review your settings to force its MSP DisplayPort into "Betaflight mode".
I hope I'll be able to add support for native Ardu fonts when I find a spare flight controller to install it and test it. (apart from the two I use to test Inav and Betaflight)
Of course, since you have a running setup, you can contribute. The change is small and should be straightforward.

lida2003, 2024-10-30:
From the png comparison, I think those icon index is different from each other. And I think currently the msposd is lack of png support. 

I have donwload some from https://sneaky-fpv.gitbook.io/hd-osd-and-tools, which is quite good/nice looking icons. But it didn't work. Maybe the format is not match with the code.

Hope msposd can have this png file soon, so I want open the requestion again. Maybe some has time and ability can work on https://github.com/OpenIPC/msposd/issues/13

tipoman9, 2024-10-30:
@lida2003 I really don't get your point. What do you mean "_lack of png support_"?
I wrote you above that ardupilot can be configured to use Betaflight's fonts.
There are several reports that this working with videos posted.
Here is one from today. 
[image]

lida2003, 2024-10-30:
We have betaflight and inav png file for icons, but I didn't find ardupilot png file.

tipoman9, 2024-10-31:
@lida2003 
Do you read what I wrote above?
_Currently msposd is not tested with ardupilots "native" fonts, since the demand for this was lacking.
Several users reported that they can configure ardupilot to output osd icon index for Betaflight fonts.
I hope I'll be able to add support for native Ardu fonts when I find a spare flight controller to install it and test it. (apart from the two I use to test Inav and Betaflight)_

There is no "VTX power icon" in Betaflight fonts, so when you configure Ardupilot to use Betaflight fonts it simply takes the glyph from the location it is supposed to be in Ardu fonts I suppose.
Since this icon is so important for you, you can manually copy it in the betaflight font png.

lida2003, 2024-10-31:
Thanks for you time and patience. And it's great that we can DIY if something we need to change.

tipoman9, 2024-11-11:
@lida2003 can you test with the latest version with OSD rendered on the cam.
[image]
Only change
```MSP_OPTIONS = 4```
The rest should be
```
SERIALx_PROTOCOL = 42 
OSD_TYPE = 5
```
It works for me with the latest ardupilot version.

lida2003, 2024-11-11:
Yes(latest version or previous version), it's ok for osd rendered on cam, using betaflight fonts+ardupilot betaflight settings. I'm trying osd rendering on ground feature, as it can save video with out osd.

tipoman9, 2024-11-11:
> Yes(latest version or previous version), it's ok for osd rendered on cam, using betaflight fonts+ardupilot betaflight settings. I'm trying osd rendering on ground feature, as it can save video with out osd.
> 
> EDIT: mw icon is the png font issue. I haven't change that yet, but it's NOT that important.
@lida2003 
Just to clarify. ArduPilot has two modes for sending OSD via MSP Display port
1 ```MSP_OPTIONS = 4``` then it must be used with its specific ArduPilot fonts.
2 ```MSP_OPTIONS = 5``` it must be used with Betaflight fonts.

So far we had option 2 running, now option 1 is also available - all you need to do is change the setting and use and ardupilot font file.

lida2003, 2024-11-12:
OK, here comes my result.

[image]

Test : using [ardupilot png](https://github.com/OpenIPC/msposd/tree/main/fonts/ardupilot)
1. `cp fonts/ardupilot/*.png /usr/share/fonts/`
2. `./release/star6e/msposd  --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14550 --osd --matrix 11 --ahi 1 -r 30 -v `
3. ardupilot png initial loaded confirmed. //default MSP_OPTIONS = 0
[image]

5. All icons are good to go (as above image)
6. switch MSP_OPTIONS to 4, and reboot FC
[image]

8. some icons, such as GPS/HOME/COMPASS is NOT good
9. switch MSP_OPTION to 6, and reboot FC
[image]
*Note: I'm NOT sure what this "unused" means, maybe should be discussed on ardupilot forum.*

11. result is the same as step #6 
12. switch MSP_OPTION to 6, and reboot FC
[image]
**Anything related to telemetry?**

13. Anyway, result is the same as step #6, no improvement

==>I believe we should set [MSP_OPTIONS](https://ardupilot.org/copter/docs/parameters-Copter-stable-V4.5.7.html#MSP_OPTIONS) = 0, if using native ardupilot png files.

tipoman9, 2024-11-12:
Well, I'm not sure for ardupilot settings, I also tested with ```MSP_OPTIONS = 0``` and it was the same as ```MSP_OPTIONS = 4```
Maybe I'll change the description in the readme to set value 0 if native ardupilot fonts are to be used.

lida2003, 2024-11-12:
I just look at the OSD. MSP_OPTIONS = 4. But the above test image is NOT good.

Really confused, I think it might need ardupilot expert here.

[image]