OpenIPC/msposd #17: [Request] Porting ground station OSD to Jetson Orin Nano

lida2003, 2024-11-03:
The request is from https://github.com/OpenIPC/msposd/issues/14, refering feature on https://www.youtube.com/watch?v=niyOnrXOFfU

As currently it's on x86, the request is on jetson orin nano, but we have segmentation error: https://github.com/SnapDragonfly/msposd/tree/porting_jetson_orin

Please help!

```
daniel@daniel-nvidia:~/Work/msposd$ sudo ./release/jetson/msposd  --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v
Ver: 20241103_084840
Listen on port :127.0.0.1:14550
MSP to OSD mode!Verbose mode!Majestic width:1280,height:720
No protocol specified
Cannot open display
Loading /home/daniel/Work/msposd/release/jetson/font.png for 1000p mode
Font file res 144:13824 pages:4
Loading small size glyphs /home/daniel/Work/msposd/release/jetson/font_hd.png for 1000p mode
Glyph size:36:54 on a 53:20 matrix. Overlay 1912:1080
Create_region PixelFormat:11 Size: 1912:1080 X_Offset:8 results: -1
Segmentation fault
```

```
Starting program: /home/daniel/Work/msposd/release/jetson/msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Ver: 20241103_084840
Listen on port :127.0.0.1:14550
MSP to OSD mode!Verbose mode!Majestic width:1280,height:720
No protocol specified
Cannot open display
Loading /home/daniel/Work/msposd/release/jetson/font.png for 1000p mode
Font file res 144:13824 pages:4
Loading small size glyphs /home/daniel/Work/msposd/release/jetson/font_hd.png for 1000p mode
Glyph size:36:54 on a 53:20 matrix. Overlay 1912:1080
Create_region PixelFormat:11 Size: 1912:1080 X_Offset:8 results: -1

Program received signal SIGSEGV, Segmentation fault.
0x0000fffff7edefe0 in cairo_set_source_rgba () from /lib/aarch64-linux-gnu/libcairo.so.2
(gdb) bt
#0  0x0000fffff7edefe0 in cairo_set_source_rgba () from /lib/aarch64-linux-gnu/libcairo.so.2
#1  0x0000aaaaaaaa7438 in Render_x86 (rgbaData=0xfffff5e49010 "", u32Width=1912, u32Height=1080) at osd/util/Render_x86.c:178
#2  0x0000aaaaaaaae480 in InitMSPHook () at osd.c:1986
#3  0x0000aaaaaaab1654 in main (argc=11, argv=0xfffffffff4a8) at msposd.c:1274
(gdb) Quit
```

tipoman9, 2024-11-04:
Sorry, can't help much, never investigated that board, I'm not sure whether and how X11 is supported there. You can check that all init calls are successful in Init_x86(...)

lida2003, 2024-11-09:
OK, I have got it running on the GCS. But I have met an issue.

- hardware definitly OK, set mavlink v2 gcs software can communicate with FC.
- hardware definitly OK, kill mavfwd and launch `msposd  --master /dev/ttyS2 --baudrate 115200 -c 7 -c 9 --out 127.0.0.1:14555 -osd -r 20 --ahi 1 -v`,  OSD on airunit works

Now, try osd on computer:
- I killed mavfwd process, started `msposd --channels 7 --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14550 --matrix 11 --ahi 3 -r 30` on airunit
- `756 root     telemetry_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0` works fine
- launch `msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v` on ground station, stucked as no data coming.
- check `wfb-cli gs`, there is no packet received from airunit.

BTW video link is always good, so there is no issue with wifi link.

Is there anything wrong with my commands?

```
root@openipc-ssc30kq:~# msposd --channels 7 --master /dev/ttyS2 --baudrate 115200 -
-out 127.0.0.1:14550 --matrix 11 -r 30
Ver: 20241027_163505
Monitoring RC channel 7
Listening on /dev/ttyS2...
Packets:3  Bytes:48
Packets:13  Bytes:210
Packets:23  Bytes:434
Packets:33  Bytes:657
Packets:43  Bytes:906
Packets:53  Bytes:1152
Packets:63  Bytes:1312
Packets:73  Bytes:1557
Packets:83  Bytes:1774
Packets:93  Bytes:1934
Mavlink buffer overflowed! Packed lost!
Packets:103  Bytes:2149
Packets:113  Bytes:2309
Packets:123  Bytes:2512
Packets:133  Bytes:2674
Packets:143  Bytes:2870
Packets:153  Bytes:3030
Packets:163  Bytes:3232
Packets:173  Bytes:3392
Packets:183  Bytes:3595
Packets:193  Bytes:3792
Packets:203  Bytes:3960
Mavlink buffer overflowed! Packed lost!
Packets:213  Bytes:4158
Packets:223  Bytes:4368
Packets:233  Bytes:4565
Packets:243  Bytes:4725
Packets:253  Bytes:4890
Packets:263  Bytes:5066
Packets:273  Bytes:5226
Packets:283  Bytes:5406
Packets:293  Bytes:5566
Packets:303  Bytes:5730
Packets:313  Bytes:5891
Mavlink buffer overflowed! Packed lost!
Packets:323  Bytes:6078
Packets:333  Bytes:6238
Packets:343  Bytes:6457
Packets:353  Bytes:6617
Packets:363  Bytes:6780
Packets:373  Bytes:6940
Packets:383  Bytes:7112
Packets:393  Bytes:7280
Packets:403  Bytes:7440
Packets:413  Bytes:7618
Packets:423  Bytes:7778
Packets:433  Bytes:7972
Mavlink buffer overflowed! Packed lost!
Packets:443  Bytes:8203
Packets:453  Bytes:8364
Packets:463  Bytes:8593
Packets:473  Bytes:8753
Packets:483  Bytes:8972
Packets:493  Bytes:9133
Packets:503  Bytes:9354
Packets:513  Bytes:9514
Packets:523  Bytes:9734
Packets:533  Bytes:9894
Mavlink buffer overflowed! Packed lost!
Packets:543  Bytes:10121
Packets:553  Bytes:10344
Packets:563  Bytes:10527
Packets:573  Bytes:10687
Packets:583  Bytes:10861
Packets:593  Bytes:11043
Packets:603  Bytes:11204
Packets:613  Bytes:11380
Packets:623  Bytes:11540
Packets:633  Bytes:11714
Packets:643  Bytes:11874
Mavlink buffer overflowed! Packed lost!
Packets:653  Bytes:12048
Packets:663  Bytes:12208
Packets:673  Bytes:12381
Packets:683  Bytes:12558
Packets:693  Bytes:12718
Packets:703  Bytes:12906
Packets:713  Bytes:13066
Packets:723  Bytes:13235
Packets:733  Bytes:13395
Packets:743  Bytes:13560
Packets:753  Bytes:13720
Packets:763  Bytes:13880
Mavlink buffer overflowed! Packed lost!
Packets:773  Bytes:14040
Packets:783  Bytes:14200
Packets:793  Bytes:14362
Packets:803  Bytes:14522
Packets:813  Bytes:14682
Packets:823  Bytes:14842
Packets:833  Bytes:15065
Packets:843  Bytes:15235
```

I can't get packet on ground station.

lida2003, 2024-11-10:
I don't know why `ParseMSP` will be false.

https://github.com/OpenIPC/msposd/blob/8e32165bd05fe1b231853ebc6b060800dbd0874f/msposd.c#L767-L775

And it seems like mavlink message???? But settings are `MSP_OPTIONS = 6` `OSD_TYPE = 5` `SERIALx_PROTOCOL=42 //displayport`.

https://github.com/OpenIPC/msposd/blob/8e32165bd05fe1b231853ebc6b060800dbd0874f/msposd.c#L776-L792

lida2003, 2024-11-10:
Packet is coming...(latest code), but the icons are not updating....

```bash
root@openipc-ssc30kq:~# ./msposd --channels 7 --master /dev/ttyS2 --baudrate 115200
--out 127.0.0.1:14550 --matrix 11 --ahi 3 -r 30
Ver: 20241110_132151
Monitoring RC channel 7
Listen on port :/dev/ttyS2
Majestic width:640,height:480
Matrix size not supported on resolutions smaller than 1920x1080p!
Glyph size:24:36 on a 50:18 matrix. Overlay 1200:72
Listening UART on /dev/ttyS2...
Flight Controller detected: BTFL
```

```bash
daniel@daniel-nvidia:~/Work/msposd/release/jetson$ ./msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v
Ver: 20241110_132343
Listen on port :127.0.0.1:14550
MSP to OSD mode!Verbose mode!Majestic width:640,height:480
Loading /home/daniel/Work/msposd/release/jetson/font.png for 1080p mode
Font file res 144:13824 pages:4
Loading small size glyphs /home/daniel/Work/msposd/release/jetson/font_hd.png for 1080p mode
Glyph size:36:54 on a 53:20 matrix. Overlay 1912:1080
Create_region PixelFormat:11 Size: 1912:1080 X_Offset:8 results: -1
set_LOGO with u32Height:1080 enPixelFormat 11
Listening UDP on 127.0.0.1:14550...
UART Events:0 MessagesTTL:0 AttitMSGs:0(0ms) Bytes/S:0 FPS:1 of 0 (skipped:0), AvgFrameLoad ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
                                                                                  FC set mode:0
FC set mode:0
FC set mode:0
$M<Flight Controller detected: BTFL
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
UART Events:11 MessagesTTL:170 AttitMSGs:4(3357926ms) Bytes/S:2650 FPS:1 of 0 (skipped:0), AvgFrameLoad ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 255| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
                                                                                   FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
UART Events:11 MessagesTTL:172 AttitMSGs:4(3358105ms) Bytes/S:2650 FPS:1 of 0 (skipped:0), AvgFrameLoad ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 2301| 00| 791| 27972| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
                                                                                        FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
FC set mode:0
UART Events:11 MessagesTTL:170 AttitMSGs:3(4197888ms) Bytes/S:2650 FPS:1 of 0 (skipped:0), AvgFrameLoad ms:0 | 0 | 0 |
Channels :| 00| 00| 00| 00| 2301| 00| 791| 27972| 00| 00| 00| 00| 00| 00| 00| 00| 00| 00
                                                                                        FC set mode:0
FC set mode:0
FC set mode:0
```

[image]
[image]

tipoman9, 2024-11-10:
@lida2003  Your setup seems ok, messages are received on the ground.
There was a problem with Betaflight OSD ground mode, just fixed it. 
Try the latest version, hope it will fix the problem for ardupilot also.

lida2003, 2024-11-11:
@tipoman9 I got both camera and jetson upgraded to latest code.

Test A: camera OSD  ==> OK
- kill `mavfwd`
- command: `./msposd --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14550 --osd --matrix 11 --ahi 1 -r 30 -v`

Test B: jetson OSD ==> NG
- kill `mavfwd`
- camera command: `./msposd --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14550 --matrix 11 --ahi 1 -r 30 -v`
- jetson command: `./msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 1 --matrix 11 -v`
- camera log:  [camera.txt](https://github.com/user-attachments/files/17694176/camera.txt)
- jetson log:  [jetson.txt](https://github.com/user-attachments/files/17694178/jetson.txt)

tipoman9, 2024-11-11:
@lida2003  
I tested the x86 version with ardupilot and it works as it should, both when rendered on the ground or in the air unit.
ArduPilot has two modes for sending OSD via MSP Display port
1 ```MSP_OPTIONS = 4``` then it must be used with its specific ArduPilot fonts.
2 ```MSP_OPTIONS = 5``` it must be used with Betaflight fonts.
Tested both of them, worked on the ground or on the camera.
Here is an icon that is present only in Ardupilot font, rendered on the ground 
[image]

Use the latest revision and check your wfb-connection - the number of bytes sent by the air unit should be about the same as the bytes received on the ground.

I would suggest to connect the FC directly to the ground station via UART, and then run on the ground only:
```msposd --master /dev/ttyUSB0 --baudrate 115200  --osd --matrix 11 --ahi 1 -r 30 -v```
This should show OSD

Then you can test the client-server only on the ground again, you can do it on a x86 first, by starting two instances of msposd
```
#on GS read serial and forward to test air no OSD mode
msposd_x86 --channels 7 --master /dev/ttyUSB0 --baudrate 115200  --out 127.0.0.1:14550  --matrix 11 --ahi 3 -r 50 -v

#Render on GS
msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v
```
Don't forget to place the appropriate font files in the same folder.
[image]

lida2003, 2024-11-12:
This is really odd.

Loopback jetson test:
1. connected FC by USB serial adapter
2. wfb-ng is down, then 14550 is free
3. launch simulated airunit `./msposd --channels 7 --master /dev/ttyUSB0 --baudrate 115200  --out 127.0.0.1:14550  --matrix 11 --ahi 3 -r 50 -v`
4. lanuch `./msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v` on jetson
5. It's all good

Using camera+jetson:
1. connect FC back to camera `/dev/ttyS2`
2. launch simulated airunit `./msposd --channels 7 --master /dev/ttyS2 --baudrate 115200  --out 127.0.0.1:14550  --matrix 11 --ahi 3 -r 50 -v`
3. start wfb-ng
4. video/mavlink are coming in `wfb-cli gs`
5.  lanuch `./msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v` on jetson
6. NG

It seems something to do with wfb-ng????

I have killed mavfwd in rc script.

```
root@openipc-ssc30kq:~# ps
PID   USER     COMMAND
    1 root     init
    2 root     [kthreadd]
    3 root     [ksoftirqd/0]
    5 root     [kworker/0:0H]
    7 root     [rcu_preempt]
    8 root     [rcu_sched]
    9 root     [rcu_bh]
   10 root     [migration/0]
   11 root     [lru-add-drain]
   12 root     [watchdog/0]
   13 root     [cpuhp/0]
   14 root     [cpuhp/1]
   15 root     [watchdog/1]
   16 root     [migration/1]
   17 root     [ksoftirqd/1]
   18 root     [kworker/1:0]
   19 root     [kworker/1:0H]
   20 root     [kdevtmpfs]
   21 root     [netns]
  185 root     [oom_reaper]
  186 root     [writeback]
  188 root     [kcompactd0]
  189 root     [crypto]
  190 root     [bioset]
  192 root     [kblockd]
  222 root     [watchdogd]
  247 root     [kswapd0]
  248 root     [vmstat]
  336 root     [kapmd]
  339 root     [bioset]
  342 root     [bioset]
  358 root     [ehci_monitor]
  361 root     [kworker/0:1]
  373 root     [urdma_tx_thread]
  401 root     [bioset]
  406 root     [bioset]
  409 root     [kworker/0:2]
  412 root     [bioset]
  417 root     [bioset]
  422 root     [bioset]
  433 root     [monitor_temp]
  440 root     [spi0]
  444 root     [spi1]
  465 root     [kworker/0:1H]
  479 root     [jffs2_gcd_mtd4]
  510 root     syslogd -n -C64 -t
  514 root     /sbin/klogd -n
  580 root     udhcpc -x hostname:openipc-ssc30kq -A 0 -T 1 -t 5 -R -b -O search -p /var/run/udhcpc.eth0.pid -i eth0
  585 root     ntpd -n
  590 root     dropbear -R -B -k -p 22 -K 300
  600 root     dropbear -R -B -k -p 22 -K 300
  608 root     [SensorIfThreadW]
  616 root     [IspDriverThread]
  620 root     [IspMidThreadWq]
  632 root     [kworker/u4:15]
  638 root     [mi_log]
  685 root     majestic -s
  700 root     [cfg80211]
  723 root     [RTW_CMD_THREAD]
  738 root     wfb_tx -p 0 -u 5600 -R 456000 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 8 -n 12 -T 0 -i 7669206 -f data
  741 root     [vif0_P0_MAIN]
  742 root     [vif1_P0_MAIN]
  743 root     [vif2_P0_MAIN]
  745 root     [vpe0_P0_MAIN]
  746 root     [vpe0_P1_MAIN]
  747 root     [vpe0_P2_MAIN]
  748 root     [vpe0_P3_MAIN]
  749 root     [VEP_DumpTaskThr]
  756 root     telemetry_rx -p 144 -u 14551 -K /etc/drone.key -i 7669206 wlan0
  757 root     telemetry_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wla
  761 root     [venc0_P0_MAIN]
  762 root     [venc1_P0_MAIN]
  767 root     /sbin/getty -L console 0 vt100
  777 root     -sh
 2232 root     [kworker/u4:0]
 2233 root     [kworker/u4:1]
 2235 root     ps
```

lida2003, 2024-11-12:
There is definitly something wrong. 
a) I only got MAVLinkV2 message when I start wfb-ng no matter if the FC is connected.
b) If I connect FC then I can see packet received from `wfb-cli gs`; If I disconnect FC then there is no packet received.

It seems the wfb-ng is blocking the message from msposd. @tipoman9  Did you met this before? 

```
daniel@daniel-nvidia:~/Work/msposd/release/jetson$ ./a.out 
UDP server is listening on port 14550...
Received message length: 20 bytes
Data in hex: FD 08 00 00 D3 03 44 6D 00 00 00 00 02 00 E5 E5 64 CC CC 39 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D4 03 44 6D 00 00 00 00 03 00 E6 E6 64 CD 07 77 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D5 03 44 6D 00 00 00 00 03 00 E5 E5 64 CD 2C 07 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D6 03 44 6D 00 00 00 00 07 00 E6 E6 64 CD 25 88 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D7 03 44 6D 00 00 00 00 03 00 E6 E6 64 CE 11 85 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D8 03 44 6D 00 00 00 00 04 00 E6 E6 64 CE BC E3 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 D9 03 44 6D 00 00 00 00 05 00 E6 E6 64 CE 43 34 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 DA 03 44 6D 00 00 00 00 03 00 E6 E6 64 CE F0 B4 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 DB 03 44 6D 00 00 00 00 02 00 E6 E6 64 CD 67 49 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 DC 03 44 6D 00 00 00 00 05 00 E6 E6 64 CD B8 7E 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 DD 03 44 6D 00 00 00 00 01 00 E5 E5 64 CC 3D 78 
Client IP: 127.0.0.1, Port: 55498

Received message length: 20 bytes
Data in hex: FD 08 00 00 DE 03 44 6D 00 00 00 00 04 00 E5 E5 64 CD 38 49 
Client IP: 127.0.0.1, Port: 55498
```

tipoman9, 2024-11-12:
I think wfb-ng server assumes by default that port 14550 is used for mavlink . Can you try to setup the msposd connection on some other port?
Or you can try to start separate instances of wfb_rx (without the wfb-ng server) like this:
```sudo ./wfb_rx -p 16 -i 7669206 -u 14550 -K /etc/gs.key  $WLANs_string```

lida2003, 2024-11-12:
> Can you try to setup the msposd connection on some other port?

Yeah, I'll try later. But still confused, maybe I'll get answer here https://github.com/svpcom/wfb-ng/discussions/383

lida2003, 2024-11-13:
With a seperate port 14560 for msposd, it works perfectly on jetson.

```bash
//airunit
/root/msposd --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14560 --matrix 11 --ahi 1 -r 30
wfb_tx -p 17 -u 14560 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0

//gs
sudo ./wfb_rx -p 17 -i 7669206 -u 14560 -K /etc/gs.key  wlan1
./msposd --master 127.0.0.1:14560  --osd -r 50 --ahi 1 --matrix 11 -v
```

tipoman9, 2024-11-13:
Just tested it, works on x86.
on the cam
```echo "Custom Message... &L20 &F28 CPU:&C &B temp:&T" >/tmp/MSPOSD.msg```
[image]

The message is sent as a new DisplayPort message type and rendered on the ground - so you can't see it in the video, you need OSD.

lida2003, 2024-11-13:
@tipoman9 

Here are three points I want to discuss here:

QA1: It works on my jetson, please check.

Jetson port: https://github.com/OpenIPC/msposd/pull/23

QA2: Please try my messge which locate at the bottom of the image, it seems NOT there(but it's OK with camera OSD)

- `echo "Custom Message... &L20 &F28 CPU:&C &B temp:&T" >/tmp/MSPOSD.msg`  ==> OK
- `echo "Monitor... &L24 &F28 CPU:&C &B temp:&T" >/tmp/MSPOSD.msg` ==> NG, maybe out of the screen

QA3: Well, please help to check the RC stick, it's NOT centered in the box

[image]

lida2003, 2024-11-14:
> QA2: seems your ground resolution is 720p or 768p. Tested only on 1080p

Yes, I set majestic 1280x720 resolution. 
And actually, I have issue with 1080P. 
- I checked the msposd binary and png file with md5sum, all seems ok.  
- 1280x720 camera osd is OK. I will be mess when I changed to the resolution 1920x1080.
[image]

lida2003, 2024-11-14:
I saw the message in saved mp4 file. Then just corrupted icons.

[image]

tipoman9, 2024-11-14:
So you get corrupted icons on 1080p and do not get them on 768p ?
If so, the issue us probably with bmp rowstride not correctly rounded for the CPU. 
Will tell you which line to change and test later

lida2003, 2024-11-14:
Now it's quite difficult to match the OSD(Full screen, not 1920x1080, slightly smaller and not centered) and video(might choose 1366x768, as my display is 1920x1080)

tipoman9, 2024-11-14:
run without ` -v` argument, this will make the window borderless  without the title bar and you will the bottom line.
But be prepared that you may need some hotkey predefined to kill msposd in such case to take controls or the system back :)
Are there any other problems? What did you change to make the icons look garbled?

lida2003, 2024-11-14:
> Are there any other problems? 

Not yet.

> What did you change to make the icons look garbled?

change majestic cfg from 1280x720 to  1920x1080

lida2003, 2024-11-16:
> QA4: https://github.com/tipoman9/wfb-stabilizer

I saw:
1. python osd???? in osd_overlay.py
2. wmctrl for overlay correction in render_direct.py
3. maybe other....

Which one do you suggest. I just got jetson working with video stabilize.

tipoman9, 2024-11-16:
I use it as my ground station "wrapper".
I starts the tools for video rendering + OSD (msposd or qopenhd) + radio link statistics.
When I need direct video rendering, I start 
```python3 /home/home/src/wfb-stabilizer/render_direct.py msposd```
when I need stabilization (+30ms latency and limited to 50fps) I start
```python3 /home/home/src/wfb-stabilizer/ejo_wfb_stabilizer.py msposd```

lida2003, 2024-11-17:
Do we really have 50FPS video stream when the delay(code excution) is 30ms?

I'm NOT cuda export, but the opencv 4.9.0 support cuda on jetson. Here is a stable comparison code block delay from image read to cv2.show. It will be stable around 20ms, which supports 50FPS video stream (20x50 = 1000). But my camera is 60FPS, then what for the other 10PFS data???? How to handle those extra video frames???? 

I think it will cause some issue as the frame is cached in the buffer, as stabilizer can't handle larger than 50FPS(20ms) frame data, am I right? what can we do about this?

[image]

lida2003, 2024-11-21:
I think the issue is closed with https://github.com/OpenIPC/msposd/pull/23.

But the stabilzer is NOT that good as I thought. Maybe I have to dig deeper.

closed 2024-11-21 by lida2003