OpenIPC/msposd #18: Can't launch msposd on Groud stattion NVR Hi3536dv100

Warfull, 2024-11-04:
It seems the msposd file in the release folder is not supported by the hardware because even if the file is present in the /usr/bin folder the system answer "file not found".

tipoman9, 2024-11-04:
Font file location has been changes to /usr/share/fonts/
Post the full log please.
I kind of don't get the idea of running msposd on the ground on NVR

Warfull, 2024-11-06:
Ok, I'm dumb : msposd is not needed on the ground station. But the result is a bit weird :
[image]

lida2003, 2024-11-06:
> msposd is not needed on the ground station.

That depends on you requirement.

Airunit can do the OSD stuff, you can take a look at this https://github.com/OpenIPC/msposd/issues/13

Warfull, 2024-11-07:
It's working now. Issue was, I guess, a bad parameter passed to the proc in the telemetry file.

Warfull, 2024-11-07:
OSD_TYPE=5
SERIALx_PROTOCOL=42
MSP_OPTIONS=4

I use Betaflight fonts.
 same result with MS_OPTIONS=6

Warfull, 2024-11-07:
But no way to get the AHI working, it is not drawn.

tipoman9, 2024-11-08:
AHI is not supported currently camera side on Hisilicon/goke. Not hard to port, if someone is willing to change the code a little and test it, a PR is welcomed

Warfull, 2024-11-11:
I'de like to try to change the code but the big issue for me is cross-compilation. I've tried to do it by the way of Buildroot and OpenIPC firmware, adding package to my board in buildroot but the compilation stop because of "BR_COMPILER_PARANOID_USAFE_PATH" ; ie "Error : Unsafe Path/Library used in cross-compilation : -I/usr/include".
I don't know how to manually do the cross-compilation.

lida2003, 2024-11-11:
@Warfull Right now, I'm stucked at arduRover https://github.com/OpenIPC/msposd/issues/17 , and I think I might do the job if I get ArduRover done.

I have 3536DV100 also, having sata harddisk connected for DVR recording.

tipoman9, 2024-11-11:
> I'de like to try to change the code but the big issue for me is cross-compilation. I've tried to do it by the way of Buildroot and OpenIPC firmware, adding package to my board in buildroot but the [...]
@Warfull 
This is a known issue at the moment, the latest code does not compile, you may try with older version of msposd till we fix it.
https://github.com/OpenIPC/firmware/commit/b3ab67b772b66c7367e7e645f1a97ae85cd9b241

Or try a standalone install, it should download the toolchain required and do the crosscompilation.
```.\build.sh goke```

Warfull, 2024-11-11:
@tipoman9 
Thank you, I've just tried and it compile. Right now I'm working on the case and hardware but I will dig into the code as soon as I can.

lida2003, 2024-11-21:
@tipoman9 latest goke version got segfault: `
```
# ./msposd_goke --master 127.0.0.1:14560 --osd -r 50 --ahi 1 --matrix 11
Segmentation fault
```

I also ported a 3536 version which uses same gcc as rootfs, still got segfault.
detailed 3536 branch info: https://github.com/SnapDragonfly/msposd/tree/porting_hi3536

tipoman9, 2024-11-21:
@lida2003 Fixed it

lida2003, 2024-11-23:
@tipoman9 I'm NOT sure what happened, it seem gcc version mis-match???
I'm using cortex_a7-gcc13-glibc-4_9 from [hi3536dv100_fpv_defconfig](https://github.com/OpenIPC/firmware/blob/master/br-ext-chip-hisilicon/configs/hi3536dv100_fpv_defconfig)

```
root@openipc-hi3536dv100:~# ls
font.png         font_hd.png      gs-lida2003.key  majestic.yaml    vtxmenu.ini
root@openipc-hi3536dv100:~# scp daniel@192.168.78.101:/home/daniel/Work/msposd/release/hisi/msposd msposd_hisi
daniel@192.168.78.101's password:
msposd                                                                                                                                100%  119KB 119.4KB/s   00:00
root@openipc-hi3536dv100:~# scp daniel@192.168.78.101:/home/daniel/Work/msposd/release/goke/msposd msposd_goke
daniel@192.168.78.101's password:
msposd                                                                                                                                100%  119KB 119.4KB/s   00:01
root@openipc-hi3536dv100:~# ls
font.png         font_hd.png      gs-lida2003.key  majestic.yaml    msposd_goke      msposd_hisi      vtxmenu.ini
root@openipc-hi3536dv100:~# ls -al msposd*
-rwxr-xr-x    1 root     root        122280 Nov 23 03:58 msposd_goke
-rwxr-xr-x    1 root     root        122280 Nov 23 03:57 msposd_hisi
root@openipc-hi3536dv100:~# ./msposd_hisi --master 127.0.0.1:14560  --osd -r 50 --ahi 1 --matrix 11 -v
-sh: ./msposd_hisi: not found
root@openipc-hi3536dv100:~# ./msposd_hisi_goke --master 127.0.0.1:14560  --osd -r 50 --ahi 1 --matrix 11 -v
-sh: ./msposd_hisi_goke: not found
root@openipc-hi3536dv100:~# file --version
-sh: file: not found
```

- jetson/x86
```
daniel@daniel-nvidia:~/Work/msposd/release$ git log -n 1
commit c1ffa361f18afd796bbb2874fa857517976a1121 (HEAD -> main, origin/main, origin/HEAD)
Author: Tipo Man <tipoman9@gmail.com>
Date:   Thu Nov 21 23:32:01 2024 +0200

    Goke/Hisi compile&binaries fixed. Commit version added in binary
daniel@daniel-nvidia:~/Work/msposd/release$ git status
On branch main
Your branch is up to date with 'origin/main'.

nothing added to commit but untracked files present (use "git add" to track)
daniel@daniel-nvidia:~/Work/msposd/release$ file hisi/msposd
hisi/msposd: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-arm.so.1, stripped
daniel@daniel-nvidia:~/Work/msposd/release$ file goke/msposd
goke/msposd: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-arm.so.1, stripped
```

lida2003, 2024-11-23:
Here is a simple cortex_a7-gcc13-glibc-4_9 substitue cortex_a7_thumb2-gcc13-musl-4_9 log:

- [CC=cortex_a7-gcc13-glibc-4_9-test.txt](https://github.com/user-attachments/files/17878439/CC.cortex_a7-gcc13-glibc-4_9-test.txt)

lida2003, 2024-11-25:
Yes, OpenIPC/firmware [change toolchain naming rule](https://github.com/OpenIPC/firmware/commit/d6c4b42bcc46cfb61229865a91d391e3be8092bd#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52R35): 

```
	@$(PWD)/general/scripts/show_toolchains.sh $(CONFIG)
	@echo br-sdk-$(BR2_OPENIPC_SOC_VENDOR)-$(BR2_OPENIPC_SOC_FAMILY)
```

And I didn't see 3536 support here (https://github.com/OpenIPC/msposd/commit/8a82c6584c6c68214f49217c74878aa9b7308e43), right? 

Previously:
- OpenIPC/builder do NOT have 3536 target.
- OpenIPC/firmware has below tagets:
> br-ext-chip-hisilicon/configs/hi3536dv100_lite_defconfig
cortex_a7_thumb2-gcc13-musl-4_9

>br-ext-chip-hisilicon/configs/hi3536dv100_fpv_defconfig
cortex_a7-gcc13-glibc-4_9

tipoman9, 2024-11-25:
Well, no idea about 3536 support, maybe add an issue in OpenIPC/firmware ?

lida2003, 2024-11-25:
It has [toolchain.hisilicon-hi3536dv100.tgz](https://github.com/OpenIPC/firmware/releases/download/toolchain/toolchain.hisilicon-hi3536dv100.tgz). I can try it tomorrow. Hope it will work as a gs OSD.

lida2003, 2024-11-28:
Hi @henkwiedig Do you have any info on Hi3536?

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/.github/workflows/firmware.yml#L25-L26

lida2003, 2024-11-30:
> In the latest firmware version when building for `hi3536dv100_fpv` msposd is _not_ build.

It should build, as there is just a dynamic library symbol issue. I'm NOT sure if it's connected to segfault. But there is something wrong with [`libmpi.so`](https://github.com/OpenIPC/firmware/tree/master/general/package/hisilicon-osdrv-hi3536dv100/files/lib) and [sdk/hi3536dv100](https://github.com/OpenIPC/msposd/tree/main/sdk/hi3536dv100/include).

I think it's from third party or from HISI, but I don't know where to ask for help. Although I have fired a issue here: https://github.com/OpenIPC/firmware/issues/1621

@tipoman9 Did you get your Hi3536 working?

tipoman9, 2024-11-30:
@lida2003 Sorry, I don't have hi3536 to test.

lida2003, 2025-01-26:
@viktorxda I have the Hi3536 board, but right now, I just found below issues. 

If there is only `PIXEL_FORMAT_ARGB ` missing, I can try to figure out and try to fix what msposd is missing.

Can you direct me to the right course?

> > In the latest firmware version when building for `hi3536dv100_fpv` msposd is _not_ build.
> 
> It should build, as there is just a dynamic library symbol issue. I'm NOT sure if it's connected to segfault. But there is something wrong with [`libmpi.so`](https://github.com/OpenIPC/firmware/tree/master/general/package/hisilicon-osdrv-hi3536dv100/files/lib) and [sdk/hi3536dv100](https://github.com/OpenIPC/msposd/tree/main/sdk/hi3536dv100/include).
> 
> I think it's from third party or from HISI, but I don't know where to ask for help. Although I have fired a issue here: [OpenIPC/firmware#1621](https://github.com/OpenIPC/firmware/issues/1621)

whoim2, 2025-02-05:
Currently msposd is not supported on nvr hi3536?

tipoman9, 2025-02-05:
@whoim2  , I don't have hi3536 to test, but based on this info:

> The region structures are different and support for PIXEL_FORMAT_ARGB is missing, this needs to be adjusted on the msposd code.

That means that almost all of the code for native OSD rendering should be ported. I don't have such NVR, neither the motivation to do it.
Of course you can have air-side rendering with it.

lida2003, 2025-02-15:
> Currently msposd is not supported on nvr hi3536?

@whoim2 It was experimental  https://github.com/OpenIPC/msposd/pull/50

I didn't have time to test, you can try and let us know if it has any problem.

lida2003, 2025-04-13:
I did a quick test on latest build on 3536 board, with root  system `master+7ed8284, 2025-04-13`

1. copy msposd_hi3536 to `/usr/bin/msposd`
2. copy `fonts/font_ardu_hd.png` to `/usr/share/fonts/font_hd.png`
3. copy `fonts/font_ardu.png` to `/usr/share/fonts/font.png`
4. start gs application with `msposd --master 127.0.0.1:14551 --osd -r 50 --ahi 0 --matrix 11`

And I met following error:

```
root@openipc-hi3536dv100:~# msposd --master 127.0.0.1:14551 --osd -r 50 --ahi 0 --matrix 11 &
root@openipc-hi3536dv100:~# Version: 9e19a1f, compiled at: 20250414_070836
Listen on port: 127.0.0.1:14551
MSP to OSD mode!
Could not open majestic.yaml.
Loading /usr/share/fonts/font.png for 0x0
Font file res 40:13824 pages:1
Loading small size glyphs /usr/share/fonts/font_hd.png for 0x0
Glyph size: 36:54 on a 53:20 matrix. Overlay: 1800x1000
Glyph size: 36:54 on a 53:20 matrix. Overlay: 1912x1080
[create_region:72]RGN_Create failed with 0xffffffff!
RGN_SetBitMap failed with 0xa0038005  5!
Listening UDP on 127.0.0.1:14551...
```

[msposd_test_on_3536.txt](https://github.com/user-attachments/files/19727935/msposd_test_on_3536.txt)

closed 2026-10-16 by widgetii

widgetii, 2026-10-17:
For the `RGN_Create failed with 0xffffffff` / `RGN_SetBitMap failed with 0xa0038005` on the Hi3536DV100 NVR in the last report: both are fixed on master.

- **#107:** msposd's `hi3536` build now draws on the hifb layer (`/dev/fb0`) instead of an RGN overlay. An NVR has no encoder channel to attach an overlay to. The OSD is sized to whatever resolution vdec set.
- **OpenIPC/firmware#2603:** the firmware builds and ships that target for hi3536dv100.
- **OpenIPC/firmware#2614:** the firmware's libmpi and kernel modules now come from the same SDK release. The ioctl mismatch behind `0xffffffff` is gone (details in OpenIPC/firmware#1621).

On a lab NVR running `msposd --osd` alongside vdec at 720p60, the OSD draws. Use an image built from master after OpenIPC/firmware@f182c17.