OpenIPC/msposd #22: Custom messages may cause garbled text if echoed too early

lida2003, 2024-11-12:
If I use command below in rc.local script, then the message seems to be garbled.

- `echo "Monitor... &L04 &F28 &F22 CPU:&C &B temp:&T" >/tmp/MSPOSD.msg`

So I add a cron task evey 1 minute, then it will be ok. I don't know why?

- at the very begining, just after the initial png file display.
[image]

- after 60seconds, then it's OK
[image]

tipoman9, 2024-11-12:
I don't get it, where is the garbled text?

henkwiedig, 2024-11-12:
I'm guessing it's the discrepancy between the green MSPOSD.msg wich looks a lot like the one from the Adaptive Link package and the custom white one at the bottom "Monitor .... "

@lida2003 do you use Adaptive Link ? This package updates the MSPOSG.msg constantly. You will not be able to use a custom one or change out Adaptive Link config to skip that.

lida2003, 2024-11-12:
@henkwiedig YES, I have just installed adaptive link. Thanks, I see the code updating MSPOSG.msg. That's great!

closed 2024-11-12 by lida2003