OpenIPC/msposd #26: Is it possible for OSD exit on button ESC?

lida2003, 2024-11-19:
I have tried to enable this line. But it seems doesn't working. Any way to get it work?

https://github.com/OpenIPC/msposd/blob/403c56d97b792f10a1656efdaa7e6a391599abdf/osd/util/Render_x86.c#L43

tipoman9, 2024-11-19:
Add the appropriate line also here : https://github.com/OpenIPC/msposd/blob/403c56d97b792f10a1656efdaa7e6a391599abdf/osd/util/Render_x86.c#L253

lida2003, 2024-11-19:
Tried on jetson, with below line added. All ESC/Alt+Up/Alt+Down are not working.

I checked code, it was invoked from draw_complete-->draw_screenBMP--DrawOSD(GS)-->FlushDrawing_x86-->register event_callback and grab key ESC

It seems OK. BUT not working. Is there anything I made wrong?

```
diff --git a/osd/util/Render_x86.c b/osd/util/Render_x86.c
index 4bc010c..622a0d2 100644
--- a/osd/util/Render_x86.c
+++ b/osd/util/Render_x86.c
@@ -40,7 +40,7 @@ void handle_key_press(XEvent *event) {
     // Print the key that was pressed
     if (keysym == XK_Escape) {
         printf("Escape key pressed, exiting...\n");
-        //exit(0);  // Exit the program on Escape key
+        exit(0);  // Exit the program on Escape key
     }
     if ((event->xkey.state & Mod1Mask) && (keysym == XK_Up)) {
         if (AHI_TiltY<300)
@@ -262,12 +262,13 @@ void FlushDrawing_x86(){
         // XGrabKey(display, XKeysymToKeycode(display, XK_Left), ShiftMask, window, True,
         //      GrabModeAsync, GrabModeAsync); // Shift + Left Arrow

-
+        XGrabKey(display, XKeysymToKeycode(display, XK_Escape), 0, RootWindow, True,
+             GrabModeAsync, GrabModeAsync); // Escape key
         XGrabKey(display, XKeysymToKeycode(display, XK_Up), Mod1Mask, RootWindow, True,
              GrabModeAsync, GrabModeAsync); // Alt + Up Arrow
         XGrabKey(display, XKeysymToKeycode(display, XK_Down), Mod1Mask, RootWindow, True,
              GrabModeAsync, GrabModeAsync); // Alt + Down Arrow
-
+
     }

     // Copy work buffer to the display surface do avoid flickering
```

lida2003, 2024-11-21:
> For the task I've configured a simple action with xbindkeys , that "kills them all".

OK, I got it. Just "kill them all", haha..... Excellent!!!  I did it.  https://github.com/SnapDragonfly/jetson-fpv

closed 2024-11-21 by lida2003