OpenIPC/msposd #33: _DEBUG_x86 MACRO optimization & clarification

lida2003, 2024-11-27:
I have searched code, and it exists in two places:

May be we can optimze this:

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/osd/util/Render_x86.c#L1-L2

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/osd/util/Render_x86.c#L73-L75

lida2003, 2024-11-28:
I think it can be removed @henkwiedig . what do you think?

As we have this option:

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/msposd.c#L114

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/msposd.c#L1350-L1353

https://github.com/OpenIPC/msposd/blob/cabc6b8ade29112f63a5a8a5621e02240d0c3a13/osd/util/Render_x86.c#L71-L72

lida2003, 2024-11-29:
Please review https://github.com/OpenIPC/msposd/pull/34

henkwiedig, 2024-11-29:
closed by https://github.com/OpenIPC/msposd/pull/34

closed 2024-11-29 by henkwiedig