OpenIPC/msposd #35: [Request] OSD info save as sperate files, not in DVR

lida2003, 2024-11-30:
As we have ground station OSD, then the DVR is raw video files.

But I can't replay the video with OSD anymore. so I'm think about if it's OK to record the OSD info to a seperate file, then we can replay with the video if necessary.

Just something like [DJI OSD srt file](https://oscarliang.com/edit-dji-osd-subtitle-files-srt/).

Well, there might be a few things to be discussed, such as:

- How to sync the video with osd file
- OSD file format, standard file???
- what application can be used for replay?

Please let us know about what you think?

tipoman9, 2024-11-30:
This is straightforward .
Just record the MSP protocol messages with a timestamp/offset for every packet.
Add a simple tool to resend the recorded data by applying the offset. 
You will have you flight data replayed.
But I don't see much benefit from this:
1. Unlike commercial systems, openipc+msposd allows the user to choose whether to have the OSD embedded into the video or not. S
2.  If you need both - raw video from the cam and video with OSD, you can do screen recording to have the OSD (and stabilization) recorded as a separate stream, like I do https://www.youtube.com/watch?v=eJz9bSjr9PQ&lc=UgxERbATpOQ7rARZJg14AaABAg
So I have no motivation for this at the moment

lida2003, 2024-12-01:
> both - raw video from the cam and video with OSD

Can Radax do both of the recording?

henkwiedig, 2025-01-26:
looked into this, just discovered that there are tools like this https://github.com/avsaase/walksnail-osd-tool and https://github.com/shellixyz/hd_fpv_video_tool who can overlay BF recorded osd to video.
i have some questions:

1. There is a .osd and a .srt file format. .osd is DJI .srt is WS, correct ?
2. Which OSD elements do they support. How are symbols stored in there ? 
3. Does anyone have samples ?
4. How to sync video and osd recording ? Start a new osd recording on arm ?

lida2003, 2025-01-27:
> 1. There is a .osd and a .srt file format. .osd is DJI .srt is WS, correct ?

I think DJI use .srt file format, see this link: https://oscarliang.com/edit-dji-osd-subtitle-files-srt/

> 2\. Which OSD elements do they support. How are symbols stored in there ?

OSD icons should be concerned by replay tool. The raw OSD data should at least include: OSD type(bf/inav/apm)/time stamp/icon index + value/icon position(option), Maybe more that I'm NOT aware of

> 3\. Does anyone have samples ?

bf black box uses mp4 video file and blackbox log.

I don't have digital goggles, and I think it would be easy to get video and osd file from DJI airunit/goggles.

> 4\. How to sync video and osd recording ? Start a new osd recording on arm ?

betaflight blackbox using kind of spot feature to sync the video and OSD.

I used to use arm osd signal/sound of beeping/rotation of the propeller to sync OSD and video in blackbox.

Here is a much more detailed link: https://oscarliang.com/blackbox/#Syncing-Video-to-Blackbox-Log

closed 2025-03-02 by henkwiedig

PetruSoroaga, 2025-12-12:
Do you have a link to github code that generated the .OSD file?