OpenIPC/msposd #41: Error copyRectI4 from 0:1944  to 1416:864  srcWidth:24, srcHeight:9216, destWidth:1280, destHeight:720

husanboyqodirov, 2024-12-18:
I'm using Ardupilot. When I execute msposd on OpenIPC, the OSD characters are blinking frequently on the video and I see this kind of message of console:
............
Error copyRectI4 from 0:1944  to 1416:864  srcWidth:24, srcHeight:9216, destWidth:1280, destHeight:720
Error copyRectI4 from 0:6048  to 1440:864  srcWidth:24, srcHeight:9216, destWidth:1280, destHeight:720
Error copyRectI4 from 0:8028  to 1464:864  srcWidth:24, srcHeight:9216, destWidth:1280, destHeight:720
............

I'm using Ardupilot fonts.

Also other OSD characters are being updated but AHI is not changing.

tipoman9, 2024-12-19:
https://github.com/OpenIPC/msposd?tab=readme-ov-file#diagnosing

husanboyqodirov, 2024-12-31:
Solved! Here's the steps:

On OpenIPC terminal type
`top`
and find the PID of mavfwd process.
Close top and
`kill {mavfwd PID}`

and then execute msposd

husanboyqodirov, 2024-12-31:
Solved! Here's the steps:

On OpenIPC terminal type
top
and find the PID of mavfwd process.
Close top and
kill {mavfwd PID}
e.g.
kill 750

and then execute msposd

closed 2024-12-31 by husanboyqodirov