OpenIPC/msposd #46: [Request] Custom message(bitrate) support for other platform except __SIGMASTAR__

lida2003, 2025-01-01:
It seems that ohter platform doesn't support bitrate custom message, am I right?

https://github.com/OpenIPC/msposd/blob/4171ce8179af06434a0e51afee72e183f6830791/osd.c#L1135-L1146

tipoman9, 2025-01-01:
The only other platform supported is HiSilicon/Goke. I'm not aware of a way to extract this type of information for it. If you find one, please share.

lida2003, 2025-01-01:
What system is our x86 compilation machine running?

Since the Pi is not supported by the build script, I'm unsure if there is a cross-build zip file available for download, as mentioned in [this issue](https://github.com/raspberrypi/tools/issues/135).

tipoman9, 2025-01-01:
msposd can run in 
1. air-unit mode. It will embed the OSD into the video using platform specific APIs. Only sigmastar and hisilicon are supported.
2. ground side mode. It will draw the OSD onto a transparent canvas, that can be placed on the foreground of the video.
The metrics you refer to is extracted only in air-unit mode from the sigmastar h265 encoder.

lida2003, 2025-01-02:
Actually, msposd has: 
- native build
- air-unit application w/o embedded OSD

And there are quite a lot of OS supported by Raspberry Pi. What would you recommend to support?

- add a build script for native build or what?

> I'm not aware of a way to extract this type of information for it. If you find one, please share.

Yes, it can be get with little coding.

```
daniel@raspberrypi:~/test $ ./test
FIFO file '/tmp/libcamera-logs' already exists.
FIFO file '/tmp/libcamera-logs' opened successfully. Waiting for input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
No more data. Waiting for new input...
..Failed(-1) to parse data. Discarding all data in the ring buffer.
...Failed(-1) to parse data. Discarding all data in the ring buffer.
...Failed(-1) to parse data. Discarding all data in the ring buffer.
...Failed(-1) to parse data. Discarding all data in the ring buffer.
...Parsed data: frame=2, fps=20.00, size=4752, bitrate=0.76 Mbps
...Parsed data: frame=8, fps=30.00, size=10752, bitrate=2.58 Mbps
.................Parsed data: frame=13, fps=30.00, size=16694, bitrate=4.01 Mbps
.....................Parsed data: frame=17, fps=30.01, size=17143, bitrate=4.12 Mbps
..................Parsed data: frame=22, fps=30.00, size=23132, bitrate=5.55 Mbps
..................Parsed data: frame=26, fps=30.00, size=15111, bitrate=3.63 Mbps
......................Parsed data: frame=31, fps=30.00, size=22758, bitrate=5.46 Mbps
.......................Parsed data: frame=36, fps=30.00, size=16483, bitrate=3.96 Mbps
........................Parsed data: frame=40, fps=30.01, size=16245, bitrate=3.90 Mbps
........................Parsed data: frame=45, fps=30.01, size=16153, bitrate=3.88 Mbps
........................Parsed data: frame=50, fps=30.01, size=8947, bitrate=2.15 Mbps
....................Parsed data: frame=55, fps=30.01, size=22814, bitrate=5.48 Mbps
....................Parsed data: frame=59, fps=30.00, size=15338, bitrate=3.68 Mbps
........................Parsed data: frame=63, fps=30.01, size=6965, bitrate=1.67 Mbps
.....................Parsed data: frame=68, fps=30.01, size=9230, bitrate=2.22 Mbps
...................Parsed data: frame=73, fps=30.01, size=12720, bitrate=3.05 Mbps
....................Parsed data: frame=77, fps=30.00, size=18207, bitrate=4.37 Mbps
.....................Parsed data: frame=82, fps=30.01, size=15290, bitrate=3.67 Mbps
.....................Parsed data: frame=87, fps=30.01, size=15234, bitrate=3.66 Mbps
```

tipoman9, 2025-01-02:
No experience with Pi and never considered supporting it, since it does not work with OpenIPC.
Feel free to experiment.

lida2003, 2025-01-03:
@tipoman9 Thanks

https://github.com/SnapDragonfly/msposd/tree/pr_pi3b_libcamera_vid_mon

[image]

closed 2025-01-03 by lida2003