OpenIPC/msposd #52: [Feature] Add SNR icon support for ardupilot

lida2003, 2025-01-30:
- The icon is included in PR https://github.com/ArduPilot/ardupilot/pull/26912 , ardupilot 4.5.5
- analog has icon, but digital don't have the icon, see below:

[image]

[image]

https://www.youtube.com/watch?v=V2TltkMQo0k

henkwiedig, 2025-02-20:
Just glaced over the Adru PR. Seem they added new glyphs to the font file.
Did you try to use the new fonts from the PR ?

lida2003, 2025-02-20:
> Did you try to use the new fonts from the PR ?

@lida2003 Sure, we have great flights using analog osd fonts since ardupilot 4.5.6.

> Looks like Sneaky didn't update his fonts as well.

I don't know Sneaky's fonts. But current msposd don't have this icon.

tipoman9, 2025-02-20:
The default fonts that ship with MSPOSD are the ones created by SneakyFPV, as mentioned in the Readme.
You can change them with other that has this glyph added.

lida2003, 2025-03-09:
I think I know where to add this glyph, but I don't know how to add even if I read some paragraph from https://sneaky-fpv.gitbook.io/hd-osd-and-tools/tools/sneaky_fpv-pilot-logo-tool

Apparently, I don't have any idea about how to use `SneakyFPV` tool or others to get the job done.

Any idea or links that can help me?

 
- [#define SYM_SNR 95](https://github.com/ArduPilot/ardupilot/blob/1f51cb395e0453b87c2f94a807b848ca2bda9718/libraries/AP_OSD/AP_OSD_Screen.cpp#L1290)
- [backend->write(x, y, blink, "%c---%c", SYMBOL(SYM_SNR), SYMBOL(SYM_DB));](https://github.com/ArduPilot/ardupilot/blob/1f51cb395e0453b87c2f94a807b848ca2bda9718/libraries/AP_OSD/AP_OSD_Screen.cpp#L2125)

lida2003, 2025-03-10:
Ah... It seems good. 

How about add gs osd multi-syle support?
@henkwiedig @tipoman9 Any suggestions?

https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_OSD/fonts/HDFonts/WalkSnail/userfont

```
.
├── user_ardu_24.png
├── user_ardu_36.png
├── WS_APC_Blinder_24.png
├── WS_APC_Blinder_36.png
├── WS_APC_Conthrax_24.png
├── WS_APC_Conthrax_36.png
├── WS_APC_Europa_24.png
├── WS_APC_Europa_36.png
├── WS_APC_Hemi_24.png
├── WS_APC_Hemi_36.png
├── WS_APC_Nexus_24.png
├── WS_APC_Nexus_36.png
├── WS_APC_Sphere_24.png
├── WS_APC_Sphere_36.png
├── WS_ARDU_Night_24.png
└── WS_ARDU_Night_36.png
```

[image]

lida2003, 2025-03-10:
> One can simply replace the default fonts per FC Software as he wishes.
> Or add a single pair of files named fond.png and font_hd.png in the folder of msposd and it will load them first, without trying to identify the FC type.

Yes, it's for kind of experienced user, who is familiar with linux shell op.

> There is not enough storage on the air side and the bmp processing is intentionally limited to 4bit colors, so the default fonts are converted from 32bit to 4bit pixel color .

Yes, I think the GS mode can be made more functional.
Thanks for point out 4bit color issue, as the png from ardupilot is kind of large, around ~200KB, msposd's png is about 30~50KB.
I didn't change pixel 4bit format, so is there any program that can convert 32bit to 4bit?

> It is better to use "original" fonts on the ground site, since they add several levels of transparency and look better - just noticed for ArduPilot they are not included in the repo.

OK, let's do step by step. Can you help to let me know where is the original ardupilot png from? Or which one should we use for orginal fonts for ardupilot, as currently on msposd there is no SNR icon.

> Maybe adding the option to switch between different fonts per FC Software with a keystroke will be a nice addition, but some convention should be used how to distinguish between different pairs of files for every FC SW...

As you mentioned above, we don't have this luxury on the air unit's NOR/NAND flash (I think). How about using an SD card? We have plenty of space on the ground station.

Then, solution might be add a path in configuration or paramter to detect multi-style font files and popup in the OSD menu, reboot the msposd, finally apply new osd font file.

Font file storage might be something like:

```
.
├── Readme.txt
├── user_ardu_24.png  //change to font_ardu_hd.png
├── user_ardu_36.png  //change to font_ardu.png
└── ardupilot
    ├── WS_APC_Blinder_24.png
    ├── WS_APC_Blinder_36.png
    ├── WS_APC_Conthrax_24.png
    ├── WS_APC_Conthrax_36.png
    ├── WS_APC_Europa_24.png
    ├── WS_APC_Europa_36.png
    ├── WS_APC_Hemi_24.png
    ├── WS_APC_Hemi_36.png
    ├── WS_APC_Nexus_24.png
    ├── WS_APC_Nexus_36.png
    ├── WS_APC_Sphere_24.png
    ├── WS_APC_Sphere_36.png
    ├── WS_ARDU_Night_24.png
    └── WS_ARDU_Night_36.png
```

lida2003, 2025-03-11:
> Yes, I think the GS mode can be made more functional. Thanks for point out 4bit color issue, as the png from ardupilot is kind of large, around ~200KB, msposd's png is about 30~50KB. I didn't change [...]

I just googled and it seems to work:
command from ImageMagick, which export 32bit png size from 203K to 52K.

```bash
$ convert font_ardu.png -depth 4 -colors 16 output.png
```

lida2003, 2025-03-29:
I have updated the latest ArduPilot PNG file, which now includes SNR ([commit link](https://github.com/SnapDragonfly/jetson-fpv/commit/18d39b2d78e2f8175f81423c37a69bf40b7c59db#diff-9fba344526ba51f5195bb7aef902d47080aa648244075dc0bce5b5614e63f9bf)).  

Since there are many PNG files, I’m not sure which one should be set as the default. Let me know what you think, and I can update it quickly once we decide on the default PNG.