OpenIPC/msposd #55: Custom message string truncation issue

lida2003, 2025-03-05:
I have this issue for a quite a log time, but I think it might be some configuration or what I have made a mess.

Can any one help to take a look: https://github.com/sickgreg/OpenIPC-Adaptive-Link/issues/15

[image]

[image]

tipoman9, 2025-03-05:
In a recent version I've added support for multiple lines (check the readme), but this works only on the air side rendering. Are you running GS rendering mode?

lida2003, 2025-03-06:
It's a BUG, I don't know why there is feedback from [Radxa don't have this issue From Mario](https://www.youtube.com/watch?v=CfqyyLivYgo). @tipoman9

lida2003, 2025-03-07:
> but this works only on the air side rendering.

Yes, it seems not working on gs side. Is there any work around?

[image]

lida2003, 2025-03-08:
[Fix custom message truncated for gs mode #56](https://github.com/OpenIPC/msposd/pull/56)

closed 2025-03-08 by lida2003