OpenIPC/msposd #74: [Feature Request] Dynamic Update of a Bar or Curve for Specific Characteristics, such as FEC

lida2003, 2025-03-29:
Currently, most OSD information consists of numbers, icons, and text messages. We propose implementing a dynamically updating graph, such as a bar or curve, to represent specific dynamic characteristics of the area.

For example, let's consider FEC (Forward Error Correction). If the FEC value is significantly high in a specific flying area, it could indicate interference, and the pilot should be aware and take action to avoid the trouble as soon as possible.

There may be several other factors to monitor as well.

I am proposing this idea/feature here for further discussion.

- [Enhancing Packet Loss Handling in FPV Scenarios with Increased FEC #21](https://github.com/OpenIPC/adaptive-link/issues/21)

tipoman9, 2025-03-29:
Yes, this is very important information. My opinion is that is not not related to MSPOSD and should be used as an external widget for the radio link framework (wfb-ng at the moment).
[This is what I use at the moment](https://github.com/tipoman9/wfb-stabilizer/blob/main/osd_overlay.py), it gets the data from a patched wfb_rx via mavlink port, so it is NOT a good solution
[image]
These are two sample screenshots and their meaning for 3 channel link  ( 3 x AR9271 adapters)
On the first image.
Channel 1 is receiving everything
Channel 2 has missed 186 packets (too much, red)
Channel 3 has missed 7 packets (not much, green)
No FEC needed, since no packet was lost (white).
On the second snapshot there is FEC involved, 10 packets recovered(yellow), since not all of the packets were received, but no packet was lost -  zero on the left.
The data is drawn over the video and over the transparent osd. 
I think similar approach should be used to create a better tool for the task.
@snokvist did a lot research on the topic and [has some nice tools for the task](https://github.com/snokvist/wfb_plotter).
Here is what I've seen from him.
[image]

henkwiedig, 2025-03-29:
Would the new pp_rk IconselectorWidget usable ?
You can select an icon based on a fact that pp_rk knows.
We could defined value rangers for lost and fec corrected packtes and draw distinct icons based on theres ranges.

lida2003, 2025-03-29:
[image]

Ruby has bars for displaying dynamic information. Is there a way to create a similar plugin that allows users to configure it easily and send relevant data via messages or other methods? The plugin would then process the data and render the bar display accordingly.